makusets/esphome-toshiba-ab #5: Protocol mismatch? Toshiba Carrier (40VD048S / MCC-1631 CN41) – need help debugging

tchlyah, 2025-08-16:
Hi @makusets, pinging @issalig as well 👋  

First of all, thanks to both of you for your amazing work on these projects.  

I’ve just received the boards and installed one on **Toshiba Carrier ducted unit** (Carrier model **40VD048S-6S-TEE**, board **MCC-1631**, using CN41 A/B).  

### The situation
- The good news: the board boots and communicates.  
- The bad news: the **protocol doesn’t seem to match** – the AC is always reported as OFF, and the only thing that seems to come through is the **remote temperature** (though not 100% accurate).  

### Logs
Here are some extracts from my session:  

<details>
<summary><b>Init</b></summary>

```text
[14:59:01][I][app:149]: ESPHome version 2025.7.5 compiled on Aug 16 2025, 14:58:38
[14:59:01][C][wifi:613]: WiFi:
[14:59:01][C][wifi:434]:   Local MAC: 40:91:51:57:17:68
[14:59:01][C][wifi:439]:   SSID: 'Keto Wifi-2.4'
[14:59:01][C][wifi:442]:   IP Address: 192.168.1.75
[14:59:01][C][wifi:446]:   BSSID: C4:E5:32:8F:C5:EB
[14:59:01][C][wifi:446]:   Hostname: 'living'
[14:59:01][C][wifi:446]:   Signal strength: -49 dB ▂▄▆█
[14:59:01][C][wifi:455]:   Channel: 1
[14:59:01][C][wifi:455]:   Subnet: 255.255.255.0
[14:59:01][C][wifi:455]:   Gateway: 192.168.1.1
[14:59:01][C][wifi:455]:   DNS1: 192.168.1.1
[14:59:01][C][wifi:455]:   DNS2: 0.0.0.0
[14:59:01][C][logger:246]: Logger:
[14:59:01][C][logger:246]:   Max Level: DEBUG
[14:59:01][C][logger:246]:   Initial Level: DEBUG
[14:59:01][C][logger:252]:   Log Baud Rate: 0
[14:59:01][C][logger:252]:   Hardware UART: UART0
[14:59:01][C][i2c.arduino:073]: I2C Bus:
[14:59:01][C][i2c.arduino:074]:   SDA Pin: GPIO2
[14:59:01][C][i2c.arduino:074]:   SCL Pin: GPIO14
[14:59:01][C][i2c.arduino:074]:   Frequency: 50000 Hz
[14:59:01][C][i2c.arduino:090]:   Recovery: bus successfully recovered
[14:59:01][I][i2c.arduino:100]: Results from bus scan:
[14:59:01][I][i2c.arduino:102]: Found no devices
[14:59:01][C][uart.arduino_esp8266:118]: UART Bus:
[14:59:01][C][uart.arduino_esp8266:119]:   TX Pin: GPIO15
[14:59:01][C][uart.arduino_esp8266:120]:   RX Pin: GPIO13
[14:59:01][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[14:59:01][C][uart.arduino_esp8266:124]:   Baud Rate: 2400 baud
[14:59:01][C][uart.arduino_esp8266:124]:   Data Bits: 8
[14:59:01][C][uart.arduino_esp8266:124]:   Parity: EVEN
[14:59:01][C][uart.arduino_esp8266:124]:   Stop bits: 1
[14:59:01][C][uart.arduino_esp8266:131]:   Using hardware serial interface.
[14:59:01][C][bme280_i2c.sensor:025]:   Address: 0x76
[14:59:01][C][bme280.sensor:181]: BME280:
[14:59:01][E][bme280.sensor:184]: Communication failed
[14:59:01][C][bme280.sensor:193]:   IIR Filter: OFF
[14:59:01][C][bme280.sensor:194]:   Update Interval: 30.0s
[14:59:01][C][bme280.sensor:196]:   Temperature 'Indoor Temperature'
[14:59:01][C][bme280.sensor:196]:     State Class: 'measurement'
[14:59:01][C][bme280.sensor:196]:     Unit of Measurement: '°C'
[14:59:01][C][bme280.sensor:196]:     Accuracy Decimals: 1
[14:59:01][C][bme280.sensor:196]:     Device Class: 'temperature'
[14:59:01][C][bme280.sensor:197]:     Oversampling: 1x
[14:59:01][C][bme280.sensor:198]:   Pressure 'Indoor Pressure'
[14:59:01][C][bme280.sensor:198]:     State Class: 'measurement'
[14:59:01][C][bme280.sensor:198]:     Unit of Measurement: 'hPa'
[14:59:01][C][bme280.sensor:198]:     Accuracy Decimals: 0
[14:59:01][C][bme280.sensor:198]:     Device Class: 'pressure'
[14:59:01][C][bme280.sensor:199]:     Oversampling: 1x
[14:59:01][C][bme280.sensor:200]:   Humidity 'Indoor Humidity'
[14:59:01][C][bme280.sensor:200]:     State Class: 'measurement'
[... 121 lines cut ...]
<details>
<summary><b>Switch from Dry to Cool</b></summary>

```text
[15:17:32][D][tcc_link.climate:083]: Enqueue command: 40:00:17:08:08:80:EF:00:2C:08:00:6A:76
[15:17:32][D][tcc_link.climate:083]: Write frame: 40:00:17:08:08:80:EF:00:2C:08:00:6A:76
[15:17:32][D][tcc_link.climate:812]: All frames written
[15:17:33][W][READER:228]: Went over buffer
[15:17:33][W][tcc_link.climate:549]: Sensor query timed out; clearing outstanding flag (timeouts=38)
[15:17:35][W][READER:228]: Went over buffer
[15:17:35][W][tcc_link.climate:781]: CRC check failed
[15:17:35][D][tcc_link.climate:083]: Failed frame: 6F:01:20:56:49:80:00:00:78:C8:0C:73:0B:01:42:61:F4:3B:61:03:01:01:00:00:5C:08:96:CE:68:26:C0:80:02:08:75:04:04:04:04:3A:A3:E4:6E:F7:B2:B8:DD:33:3A:EF:C7:B9:FB:FF:FF:FF:FF:FF:FF:FF:FE:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:F9:DE:FF:FF:FF:FF:FF:FF:FF:FF:FB:FF:FF:FF:FF:FF
 ```
</details>

### Ask
- Do you think this might be a **different dialect of TCC-Link**?  
- Any suggestions on how to adjust the protocol definitions / parser so we can make sense of these frames?  

I’d be happy to capture more logs or try changes on my side to help debug and move this forward.  

Thanks again for all your work! without your repos this wouldn’t even be possible 🙏  

Best,  
Tayeb

makusets, 2025-08-19:
Yes, occasionally the status messages from the AC are read correctly so HA should show the unit on. The protocol is certainly the same, no doubt about that. I have to investigate the issue with dropped frames. I will be doing some hardware tests as well to see if that is the cause. That's why I want you to measure the voltage. I have plenty of parts and a couple of boards with me, if it ends up being a hardware issue and I can fix it, I'll send you another one. Given the project is now public, I want it to work properly.

We will get there eventually.

tchlyah, 2025-08-20:
It looks like same as https://github.com/issalig/toshiba_air_cond/issues/55

I have around 16.2V same as @MrBlue90! BTW, have you managed to fix it?

How can we make the board more generic to accept a wide range of voltage ?

Please correct me if I'm wrong!

MrBlue90, 2025-08-20:
Unfortunately I have not been able to get it working yet. I'm trying to borrow a proper digital oscilloscope, but haven't been able to get my hand onto one yet.
For whatever dumb reason, the logic analyzer doesn't pick up any of the signals of my remote at all, so I'm glad to ready you at leasy have some success.

makusets, 2025-08-21:
I have been investigating a bit more today with my boards. I came to the conclusion that the issue comes from the noise caused by the power supply in the board. For some reason in my installation, the noise is not so bad and works for me. However, in your case, the noise is too much. I am quite sure that if you were to isolate the power supply from the board and power the board from usb, it would all work. Is there any chance you could do that?

All you would need to do is unsolder D1, near AB connector. That would cut power to the power supply and then you could power the board through usb to see if it works.

Let me know if you do that.

In the meantime I am testing a noise filtering circuit to add to the board.

issalig, 2025-08-21:
> Hi [@makusets](https://github.com/makusets), [@issalig](https://github.com/issalig),
> 
> [@issalig](https://github.com/issalig) software logs: [issalig logs.txt](https://github.com/user-attachments/files/21899453/20250820.issalig.board.txt)
I see there is communication in both sides but some noise as @makusets said. Have you been able to send commands with my code? 
If you want a more complete log plug it via serial. It will also report decoding errors.

tchlyah, 2025-08-21:
> All you would need to do is unsolder D1, near AB connector. That would cut power to the power supply and then you could power the board through usb to see if it works.

@makusets, I can't wait to test it out, but even the small tests I did yesterday was very difficult. I got COVID...

I'll do my best to do that ASAP and let you know.

> I see there is communication in both sides but some noise as @makusets said. Have you been able to send commands with my code?

@issalig, I had the exact same behavior with your board, some message went through, especially the temperature readings, but most was lost. So as @makusets mentions, this is clearly hardware issue.

Lectere, 2025-08-25:
I'm back from vacation, and about to, try to, build my own hardware. But if I understand correctly, it's better to wait a bit, because a new hardware revision is coming up right?

makusets, 2025-08-25:
> I'm back from vacation, and about to, try to, build my own hardware. But if I understand correctly, it's better to wait a bit, because a new hardware revision is coming up right?
> 
> The readme claps about a 'EasyEDA.zip' file. Cannot find that file, only a gerber.zip

Hi @Lectere . Yes, I just designed a new board that I will be testing in the coming weeks when I get them in. The current design works fine in my AC unit as intended, also works with @tchlyah Carrier unit, as long as it is powered from usb. The aim of the new design is to make it more robust and be able to work with power from AB line with other units.

I am quite happy with the firmware and I am not planning on making big changes on it.

makusets, 2025-08-29:
@tchlyah , yes, I ordered the parts and the boards some days ago, they usually take about 2 weeks to come in. I will keep you updated. I made the new board as robust as I could with my limited knowledge. It adopts a different approach to reading and sending messages which should be less prone to interference and noise. Also the power circuit was redesigned, with separate grounds for power and logic. I also added support for a future UART touchscreen and some user configurable jumpers. We will see.

makusets, 2025-08-29:
> Send me one!, I'd be happy to financially compensate royally. I'll be testing, I have a lot of experience with documentation and software development. Drop one in Skippy's pouch, and tell him to hop over here! ;)

I you give me an address, I can send one of the current boards, I have a spare one now. It might work well with the AB line power or you might need to use USB if not lucky. You just need to pay for postage if interested. Should be around 12-15 euros to EU. That's with Skippy, of course. Otherwise, you could wait some weeks and see if the new model is better.

makusets, 2025-08-29:
BTW, @tchlyah, did you try issalig code as well? What was the outcome?

tchlyah, 2025-08-29:
> [@tchlyah](https://github.com/tchlyah) , yes, I ordered the parts and the boards some days ago, they usually take about 2 weeks to come in. I will keep you updated. I made the new board as robust as [...]

Looking forward to it! I'm sure you've done an excellent job and this one will work great 😊 

> @Lectere, I you give me an address, I can send one of the current boards, I have a spare one now. It might work well with the AB line power or you might need to use USB if not lucky. You just need to pay for postage if interested. Should be around 12-15 euros to EU. That's with Skippy, of course. Otherwise, you could wait some weeks and see if the new model is better.

I just sent one to him!

> BTW, @tchlyah, did you try issalig code as well? What was the outcome?

Yes I did, I have shared before the logs, it was the same results as with yours. However I didn't try it with the power source, as I was pretty happy with the outcome with your code!

Lectere, 2025-08-29:
Tayeb send me one, so I hope I can go and start testing soon!, keep you posted!

tchlyah, 2025-09-09:
While waiting for the new board to arrive, I'm thinking about moving the board to the AC motherboard directly rather than near the thermostat.

[image]

[image]

According to the motherboard diagram, it seems that the CN41 (AB line) is the black wires near the blue ones in the right. WDYT @makusets, is it safe to connect the board directly to those, or it's better to stick to the thermostat itself ? 

In case the new board fails, what if I can use some kind of power source from the motherboard directly? WDYT ?

tchlyah, 2025-09-09:
Alright, I’ll give that a try. Although using a USB power source might complicate things! I’ll probably need an electrician for this since I’m not comfortable with that kind of installation.

Still no news of the new board?

makusets, 2025-09-16:
> Alright, I’ll give that a try. Although using a USB power source might complicate things! I’ll probably need an electrician for this since I’m not comfortable with that kind of installation.
> 
> Still no news of the new board?

@tchlyah, I received the boards and components earlier last week. The new design incorporate some adjustable components to be able to tune the board to different voltages. However, I am still having trouble with the adjustments, it works alright through USB, but I am still having trouble when powered from the AB line. I am still working on it. I will keep you updated of the progress.

tchlyah, 2025-09-27:
Hi @makusets,

So I've installed the board in the motherboard directly with USB power module, it worked perfectly in the living room, but not in the other rooms 😞 I tried changing the master id to multiple values, but nothing worked. The remote temperature is accurate, but no control is working!

Here the logs:

[logs_kitchen-ac_logs.txt](https://github.com/user-attachments/files/22571663/logs_kitchen-ac_logs.txt)

[logs_kitchen-ac_logs-2.txt](https://github.com/user-attachments/files/22571664/logs_kitchen-ac_logs-2.txt)

makusets, 2025-09-27:
Not sure what you mean @tchlyah, so the esp board is in the indoor unit, and what is it that doesn't work? The original Toshiba wall remotes? I assume you have more than one of them?

Btw, I'm away for a few weeks, so i haven't spend much time working on the new revision, I'll go back to it when back home.

Miquel

tchlyah, 2025-09-27:
I have 5 units, each one have one thermostat (or wall remote). The one that I tested first, the living room, is working perfectly. But not the others, I shared with you the logs of the Kitchen one, but all the others report the same logs.

makusets, 2025-09-28:
So, you have a ducted system yeah? 1 outdoor unit and one indoor unit in the ceiling space or similar? Or do you have 5 different ducted systems each with its own indoor unit completely independent? 
For what I can see, the kitchen remote uses master id 0x02, that's different of from 0x04 from the living room. You would need to change that for it to work.

makusets, 2025-09-28:
I just modified the code to auto detect the master address. If you don't configure a master address or if it is detected that it doesn't match the messages, it will automatically be corrected. Can you try removing the master: option from the yaml file and see what happens. I have not tested it extensively so let me know if it fails.

tchlyah, 2025-09-28:
It's a VRF system (XPower) with 5 different ducted system each with its own indoor unit and remote control.

Yes, 0x02 was my very first try, then I tested some few values including 0x40, doesn't make sens but I was desperate. The logs I shared in the message just above are the latest with 0x02. 

I have also set the correct values for the two others, same results, the board is only able to receive remote data (Ping and Temperature as you can see from the logs)

Ok I'll test the new code but I doubt it will male any difference, as I believe that the Master id is correct from the logs.