makusets/esphome-toshiba-ab #6: Can't find EASYEDA source file

yverty, 2025-08-30:
Hi, could you provide the EASYEDA file. 

I would like to change the forme factor of the PCB. 

Thanks in advance. 

BR

Julien

Lectere, 2025-08-31:
Is it not inside the zip file? (the gerber zip)

makusets, 2025-09-04:
Sorry @yverty, didn't see your message earlier. I have now uploaded the project for you to edit as you wish. I am in the process of upgrading the board to be more reliable. New version should be available in a week or two if it all works well. In case you prefer to wait.

yvertyman, 2025-09-05:
> Sorry [@yverty](https://github.com/yverty), didn't see your message earlier. I have now uploaded the project for you to edit as you wish. I am in the process of upgrading the board to be more [...]

Thanks very much for anwsering and the project. I'll wait for the new version. 

BR

closed 2025-10-05 by makusets