makusets/esphome-toshiba-ab #8: Unable to use code in custom board

makusets, 2025-09-13:
Hi @litinoveweedle,

I just openned a new issue for this to keep it tidier.

So, when you say that the board does not past the slef test, what do you mean with that? Do you have the original RBC-AMTU31-E connected too? Or are you trying to connect your board as standalone? Also, when you say it resets the AB line, what do you see? Does the toshiba RBC wall temote go blank and restart?

I will have a look at your schematic now.

Miquel

> Hello guys, I am strugling with similar already mentioned issue.
> 
> - Internal AC Unit is RAS-M16S4MUVG-E
> - Controller is RBC-AMTU31-E
> - AB voltage on scope seems to be 17V high, 15.6 low (see picture - my board not connected)
> - I designed own hybrid board between Issaligs and Makusets (see schema)
> - I discovered, that new AC units has as default TU2C protocol enabled, as never replacement for TCC. There is [service bulletin](https://www.toshiba-aircon.co.uk/wp-content/uploads/2023/03/Compatibility-Bulletin-TCC-Link-or-TU2C-Link-HM-Indoors.pdf) how to check which protocol is used and also procedure to switch it. My unit was in TU2C mode, I switched it to TCC.
> 
> Unfortunately when I connect my board to the AC unit, it does not pass some startup self test and it is impossible to switch it on. It also repeatedly resets AB line - as my board is powered from it it restarts my ESP32.
> 
> 
> [image]
> 
> [image]
> 
> Do you have any idea were could be the problem? Thank you very much. 

 _Originally posted by @litinoveweedle in [#5](https://github.com/makusets/esphome-toshiba-ab/issues/5#issuecomment-3287018388)_

makusets, 2025-09-13:
Have you tried putting 1 or 2 decent size diode in series on the A line at hte entrance to lower the voltage?

litinoveweedle, 2025-09-28:
Hello,

sorry for long pause, but I have unfortunately other tasks to attend ;-o

I do not have updates on the AC board, but I have updates on the other board for Toshiba TCC protocol controlled Heat Pump. It does use same Issalig base schema, but this time without embedded power supply circuit, therefore eliminating probable issue with it. The board is using POE powered ESP-32 (Olimex ESP32-POE board). Controlled Toshiba Heat Pump is Estia Indoor unit HWS-804XHWT9-E1 (HWS-1404WWHT9-E1), remote controller HWS-AMS11E.

The used board schema is therefore very simple:

[image]

I am using this ESPhome integration: https://github.com/vakkeli/toshiba_uart_ctrl

The issue is unfortunately same - it seems that I am unable to read anything from the TCC link, as the voltages are again above @issalig board design ranges. TCC high is around 16.8V, TCC low when set by original Toshiba controller seems to be 1V lower - around 15.8V.

This is the signal without board connected to TCC bus:

[image]

With the board connected, the TCC high drops to around 16.6V, while TCC low set by the Toshiba controller remains around 15.8V. You can once again see signal from the Toshiba controller here:

[image]

Here I captured both signal from the Toshiba controller (on the right) and probably from the board (on the left). You can clearly see differences in the signal quality - high under and overshoots but also in different TCC low levels - board set TCC low at about 1.6V bellow TCC high:

[image]

Even if I set ESPhome UART debug, the board doesn't read anything from the TCC bus, as probably higher than expected TCC levels set read optocoupler always on. This is second Toshiba device with higher TCC levels than the @issalig based board expects. 

Do you think it would be OK to use instead of the D2 in the read circuit trimmer potentiometer as voltage diver, to make the board suitable for the different voltages? With the write circuit it could probably stay same, as I thing that Toshiba devices will be OK with lower values, but It would be probably better to use similar principles where the TCC low could be made tunable. What is your opinion?

makusets, 2025-09-28:
That's very valuable info, thanks. I had the same idea, my second iteration which I was working on did actually incorporate some potentiometers to be able to adjust the voltage; however, if the voltage on the line changes again (if adding another wall remote for example, or changing the model) the problem appears again and needs to be tuned again. I'm now working on a more standard approach where the read is done after filtering the DC out, as is common in power over data lines. I think it's a more robust solution. I'm still playing with the potentiometer boards as I have them around, but not sure that's the definitive solution. When I finish the third iteration, I'll compare the results and see.
I'm confident that the solution is not far.

litinoveweedle, 2025-09-28:
When you mentioned removing DC part from the signal, do you use [bias tree](https://en.wikipedia.org/wiki/Bias_tee) or op amp?

makusets, 2025-09-29:
> When you mentioned removing DC part from the signal, do you use [bias tee](https://en.wikipedia.org/wiki/Bias_tee) or op amp?
I am using a conventional DC coupling filter followed by a comparator to have clean 3.3v logic. I tried the opamp approach in my version 2 board with the potentiometers, but it still requires adjusting to the voltage.

issalig, 2025-09-29:
Hi, I've been following your progress and it looks promising. I am thinking to make another version of my circuit but I am not an electronics engineer and I do not really know what's the best solution.

I have seen solutions using comparator LM393 and/or opamps. I am waiting for some components and once here I will start the experiments.

issalig, 2025-09-29:
This page (https://burro.hatenablog.com/entry/2022/07/18/230300) describes the schematics of the RBCAMT31 
Maybe, we can just use this circuit or get inspiration.

litinoveweedle, 2025-09-29:
Very nice ...

[image]

I can see and understand the input diode bridge to prevent polarity, the power source part and also decoupling DC part from the signal by 100uF capacitor...

makusets, 2025-09-29:
> This page (https://burro.hatenablog.com/entry/2022/07/18/230300) describes the schematics of the RBCAMT31 
> Maybe, we can just use this circuit or get inspiration.

Thanks mate, that's really useful. My second iteration used a similar design, but this should be better and tested. I'll punch some values in ltspice and see what I get.
Bear in mind that Toshiba used an LDO to power its board, which I think it's not possible to power the ESP as the heat generated would be too high. So the buck converter option available to us will introduce noise which needs to be dealt with.

litinoveweedle, 2025-09-29:
I just quickly connected my original board with the TSR DC-DC converter to the LAB PSU and oscilloscope and there is not much noise. I can image PSU has beefy capacitors on the output, but so I will test it on the TCC line as well later on. But so far it doesn't seems so bad.

[image]

makusets, 2025-10-01:
That sound good, but to be reliable you should measure it with the load (so the switching frequency is the same). The ESP draws about 30mA average with pulses in the 100mA range when using wifi. This creates pulsating current which the DCDC buck have to deal with.

makusets, 2025-10-02:
@issalig, just finished the new design, the schematics you attached were very helpful. I have ordered new boards and will test them as soon as i get them. I believe this time I got it right. I have just followed the toshiba design: comparator for the read circuit, with a DC filtering capacitor and just sink current through a 330ohm for the write circuit. We will see.

issalig, 2025-10-02:
Great news!
Regarding the different voltage levels I have seen this circuit for door entry systems that uses a digital pot MCP4017 and the does some SW tuning.
https://github.com/Elektroarzt/simplebus2-mqtt-bridge/tree/main?tab=readme-ov-file#Hardware

litinoveweedle, 2025-10-02:
@issalig I think that using a comparator solves the issues with the differences in the TCC voltages. 

@makusets is your new board powered from the TCC? Also I like using bridge rectifier as polarity protection, as TCC bus claims to be polarity insensitive. Last questions will there be any galvanic separation of TCC part from the ESP (like previously on issaligs board)?

makusets, 2025-10-02:
I haven't added the rectifier, but you are right, that should be a good idea. My board does have a diode though that would not work if plugged in rerverse. I will think about adding it if i need to change anything else. I did some simulation in LTSpice and the latest iteration that i just ordered should accept any voltage and would work with a drop of only 0.2V in the line when data is received. Regarding isolation, not the esp will not be isolated from the bus, this is not possible using the comparator approach. This is the same approach as Toshiba uses. issalig board is powered independently through usb and uses optocouplers, hence the isolation. I power my current board from the AB line, and although I still use the optocouplers at the moment, the isolation is lost as the DCDC buck has common ground. I moved awat from the optos and the only option I see is using optocouplers and an isolated DCDC buck; however, I cannot see the need for that.

makusets, 2025-10-02:
BTW, you cannot power an esp12 directly from 5V. There's no regulator to 3.3v on that board.

makusets, 2025-10-21:
@litinoveweedle, sorry for the late reply. I just finished testing the new design and it works fine with power from the AB line. If you would like to test it, I am happy to send you one. I will update the repo soon with the info.

tchlyah, 2025-10-21:
Hi @makusets,

That’s an excellent news! I would love to test the new board! Would you please send me one? I’ll be paying both the board cost and any postal cost. I’ll drop you a DM with my WhatsApp number on [Home Assistant Community](https://community.home-assistant.io/) forum.

Thank you so much!

litinoveweedle, 2025-10-21:
Hello, thanks for the great news. Would it be possible to post the connection diagram so I can update my previous PCB design? Thank you.

litinoveweedle, 2025-10-22:
@makusets, @issalig Thank you both for your insights and answers, I really appreciate your help.

4. I will try first to use only DC-DC converter AB -> 5.5V and use it for ESP-32 dev board power, utilizing its internal 3.3V rail to power rest of the board logic. I will place maximum allowed value capacitor on the DC-DC (U3) output to allow for potential ESP consumption spikes.
5. Yes, you are fully correct, I meant ESP-32 dev board.

I will try to design and order the PCB board asap and if it work publish the PCB project as well.

makusets, 2025-10-22:
> 4. I will try first to use only DC-DC converter AB -> 5.5V and use it for ESP-32 dev board power, utilizing its internal 3.3V rail to power rest of the board logic. I will place maximum allowed value capacitor on the DC-DC (U3) output to allow for potential ESP consumption spikes.
>

I think that's a good idea. I would get a dexu branded DCDC rated at 1000mA, or 2000mA even better, same footprint. These can take really big capacitors, have a look at the datasheet.

issalig, 2025-10-23:
I do not remember where I did mention the 100mA, maybe it comes from here https://burro.hatenablog.com/entry/2022/07/18/230300

Regarding ESP consumption I got this table from Google AI

<span jsuid="Ddw5tf_2x" class="uJ19be notranslate" jsaction="rcuQ6b:&amp;Ddw5tf_2x|npT2md" jscontroller="udAs2b" data-wiz-uids="Ddw5tf_2x,Ddw5tf_2y" data-complete="true"><span class="vKEkVd" data-animation-atomic="" style="" data-sae=""> </span></span>
  | ESP32 | ESP8266 | Winner
-- | -- | -- | --
Active Wi-Fi | ~160–260 mA | ~70–170 mA | ESP8266
Deep Sleep | ~10 µA | ~20 µA | ESP32
Lowest Power Mode | Hibernation (~5 µA) | Deep Sleep (~20 µA) | ESP32

litinoveweedle, 2025-10-23:
> I'm not sure what you mean with the capacitor limits. The buck that I use accepts up to 1000uF, according to the datasheet. The 1A and 2A model go as high as 4400 or so. And by using a resistor in series you could go as high as you wanted. Burrocargado goes to 1.5F in his board.

The bigger the DC-DC current output the bigger is EMI (Altough I know that so far EMI was not concerns as you mentioned), The typical DC-DC i can get hands on are something like K7803-500R3/R4 having 500mA output and max capacitance of 680uF, and that doesn't mean only the filtering cap - there is IMHO some capacitance of the ESP itself, for sure on the dev board. With such DC-DC step down suggested filtering cap would be somewhere around 500uF.

If we increase capacitance, than we need to also increase DC-DC max current (by replacing it by 1000-2000mA models) , but any such increase will also create bigger ripples during caps charging times (as more current can be drawn at time).

I don't want to over-engineer the board, but choosing low power ESP seems to be like good option. So I am almost decided for the ESP8266 (probably in the ESP12 factor - same as @makusets), the only question is if not use ESP8266 dev board instead (pros is onboard USB / TTL UART, reset+boot buttons for easy initial user programing, the cons is additional components power draw, especially when such components will not be needed most of the time).

closed 2026-04-26 by makusets